POETSII/Orchestrator #102: UIF error: 'Token (Sy_dqut[ ]) out of sequence' at Orchestrator startup

AlexRast, 2019-09-25:
Seen when running Orchestrator from manual environment configuration - same basic environment as the Orchestrate script, just entered in manually - on box Byron.

Several successful Orchestrator runs completed on Byron before this message appeared.

See screenshot for details.
[image]

mvousden, 2019-09-30:
Hey @AlexRast, I suspect this due to a change I've introduced with the launcher, but it would be useful to get some more information to narrow down the problem. Could you post the:

 - The Git revision (`git rev-parse --short HEAD`), and any modified source

 - A dump of your environment (using `env`)

Also:

 - Does the POETS prompt work after this error is printed? If so, can you command `topology /dump = file.txt`, and attach `file.txt` to this issue if it's of a reasonable size?

 - Could you (for extra points), could you `gdb` root, and step through the `Root::Root(int, char*, string)`, and determine the source that causes this output to be printed to the screen?

mvousden, 2021-06-09:
Closing, as unable to reproduce.

closed 2021-06-09 by mvousden