POETSII/Orchestrator #113: Tinsel 0.6.2 compatibility

heliosfa, 2019-10-10:
At my request, @mn416 has tweaked the semantics of `HostLink::recv()` to receive entire messages rather than individual flits. The Mothership needs to be updated to use `HostLink::recvMsg()` with the size specified as a single flit to maintain current functionality.

mn416, 2019-10-10:
Hi @heliosfa, just to clarify, the feature will available in 0.6.3.

In Tinsel, the master branch always contains the latest stable code.  When I want to introduce a significant new feature or backwards-breaking feature, I create a new release before adding the feature (so current master will be released as 0.6.2), then I put the feature into master with the next version number, which will be 0.6.3 in this case.  Master will continue to be 0.6.3, allowing bug fixes or minor tweaks, until the next major feature, when it will be "released".

heliosfa, 2019-10-10:
Thanks for the clarification!

closed 2019-10-11 by heliosfa