POETSII/Orchestrator #124: Optimised Orchestrator compilation introduces warnings

mvousden, 2020-01-28:
Compiling the Orchestrator processes using `gcc` with `O3` results a variety of warnings. `O3` significantly helps with placement runtime, and may also yield performance benefits for Mothership message brokerage. The following warnings are presented:

For gcc 7.4.0 (i.e. the one used on the POETS boxes):

```
../../Source/Launcher/Call.cpp: In function ‘int Call::call(std::vector<std::__cxx11::basic_string<char> >, std::__cxx11::string*, std::__cxx11::string*)’:
../../Source/Launcher/Call.cpp:29:9: warning: ignoring return value of ‘int pipe(int*)’, declared with attribute warn_unused_result [-Wunused-result]
     pipe(stdoutPipe);
     ~~~~^~~~~~~~~~~~
../../Source/Launcher/Call.cpp:30:9: warning: ignoring return value of ‘int pipe(int*)’, declared with attribute warn_unused_result [-Wunused-result]
     pipe(stderrPipe);
     ~~~~^~~~~~~~~~~~
../../Generics/flat.cpp: In function ‘void IError(int)’:
../../Generics/flat.cpp:380:6: warning: ignoring return value of ‘int scanf(const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
 scanf("%c",&cc);
 ~~~~~^~~~~~~~~~
../../Generics/flat.cpp: In function ‘std::__cxx11::string GetStr(FILE*)’:
../../Generics/flat.cpp:664:6: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
 fread(&len,2,1,fb);                    // Length written as 2 bytes
 ~~~~~^~~~~~~~~~~~~
../../Generics/flat.cpp:665:6: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
 fread(buf,sizeof(char),min(len,LEN)+1,fb);
 ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../Generics/flat.cpp: In function ‘void freadstr(std::__cxx11::string&, FILE*)’:
../../Generics/flat.cpp:841:6: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
 fread(&len,sizeof(unsigned),1,fp);     // Pull in the string length
 ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../Generics/flat.cpp:847:6: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
 fread(pbuf,1,len,fp);                  // Pull the characters in
 ~~~~~^~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:862:0,
                 from /usr/include/c++/7/cstdio:42,
                 from /usr/include/c++/7/ext/string_conversions.h:43,
                 from /usr/include/c++/7/bits/basic_string.h:6361,
                 from /usr/include/c++/7/string:52,
                 from ../../Generics/flat.h:6,
                 from ../../Generics/flat.cpp:3:
In function ‘size_t fread(void*, size_t, size_t, FILE*)’,
    inlined from ‘std::__cxx11::string GetStr(FILE*)’ at ../../Generics/flat.cpp:665:6:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:292:71: warning: call to ‘__fread_chk_warn’ declared with attribute warning: fread called with bigger size * nmemb than length of destination buffer
  return __fread_chk_warn (__ptr, __bos0 (__ptr), __size, __n, __stream);
                                                                       ^
../../Generics/uif.cpp: In member function ‘void UIF::pAtl(UIF::Node*)’:
../../Generics/uif.cpp:723:25: warning: ‘pExpr’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               pExpr->Add(Expr());
               ~~~~~~~~~~^~~~~~~~
../../Generics/uif.cpp:726:48: warning: ‘pVari’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               pVari->Type(No_vari); pLabl->Type(No_labl);                 break;
                                     ~~~~~~~~~~~^~~~~~~~~
../../Generics/uif.cpp:726:26: warning: ‘pLabl’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               pVari->Type(No_vari); pLabl->Type(No_labl);                 break;
               ~~~~~~~~~~~^~~~~~~~~
../../Generics/uif.cpp: In member function ‘void UIF::Add(std::__cxx11::string)’:
../../Generics/uif.cpp:225:36: warning: ‘pName’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               if (pRecd==0) CmtProc(pSect,pName);
                             ~~~~~~~^~~~~~~~~~~~~
../../Generics/uif.cpp:215:19: warning: ‘pValu’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               pQal(pValu);                                                break;
               ~~~~^~~~~~~
../../Generics/uif.cpp:230:48: warning: ‘pVari’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               pVari->Type(No_vari); pLabl->Type(No_labl);                 break;
                                     ~~~~~~~~~~~^~~~~~~~~
../../Generics/uif.cpp:230:26: warning: ‘pLabl’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               pVari->Type(No_vari); pLabl->Type(No_labl);                 break;
               ~~~~~~~~~~~^~~~~~~~~
../../Generics/uif.cpp:226:27: warning: ‘pBody’ may be used uninitialized in this function [-Wmaybe-uninitialized]
               else CmtProc(pRecd,pBody); // GCC chokes if break follows else
                    ~~~~~~~^~~~~~~~~~~~~
../../Generics/uif.cpp: In constructor ‘UIF::UIF(int, char**)’:
../../Generics/uif.cpp:96:25: warning: ‘cOp’ may be used uninitialized in this function [-Wmaybe-uninitialized]
[... 31 lines cut ...]

```
../../Generics/uif.cpp: In member function ‘void UIF::Add(std::string)’:
../../Generics/uif.cpp:225:36: warning: ‘pName’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  225 |               if (pRecd==0) CmtProc(pSect,pName);
      |                             ~~~~~~~^~~~~~~~~~~~~
../../Source/Root/Root.cpp: In member function ‘unsigned int Root::CmExit(Cli*)’:
../../Source/Root/Root.cpp:293:19: warning: ‘lIdx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  293 | Post(50,pPmap[lIdx]->M[pL],int2str(pL));     // Ensure LogServer goes last
      |                   ^
../../Source/OrchBase/HardwareFileReader/HardwareFileReaderDialect3Getters.cpp: In member function ‘bool HardwareFileReader::d3_get_validate_default_types(UIF::Node**)’:
../../Source/OrchBase/HardwareFileReader/HardwareFileReaderDialect3Getters.cpp:330:18: warning: ‘arrayIndex’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  330 |         unsigned arrayIndex;
      |                  ^~~~~~~~~~
../../Source/OrchBase/HardwareFileReader/HardwareFileReaderDialect3Sections.cpp: In member function ‘bool HardwareFileReader::d3_validate_types_define_cache()’:
../../Source/OrchBase/HardwareFileReader/HardwareFileReaderDialect3Sections.cpp:537:45: warning: ‘typeLine’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  537 |                     errors.push_back(dformat(
      |                                      ~~~~~~~^
  538 |                         "L%u: Duplicate definition of field 'type' in the "
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  539 |                         "'%s' section (previously defined at L%u).",
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  540 |                         record->pos, sectionName.c_str(), typeLine));
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

heliosfa, 2020-01-28:
Doesn't look like too much effort to fix. I'm assuming you are going to address all of the warnings across both compilers?

mvousden, 2020-01-28:
Yep (as best as I can)

PR #130 (2020-03-05): Orchestrator compiles with O3 (and pedantic)
Resolves #124. 

This changeset alters the build process for the Orchestrator to compile with a higher degree of optimisation. Compilation has been tested under GCC 7.4.0 (i.e. Cambridge hardware) and GCC 9.2.1. The 128x128 heated plate completes on Byron in around 40 seconds when the hardware model is constrained to 16 devices per thread.
closed 2020-03-05 by mvousden