POETSII/Orchestrator #175: The XML validator does not complain about edges from non-existent devices and leads to broken source generation

heliosfa, 2021-04-13:
Loading in an XML with edges coming from devices that don't exist does not throw an error and results in incorrect source generation.

The attached XML [Reactive6.xml.txt](https://github.com/POETSII/Orchestrator/files/6305262/Reactive6.xml.txt) has four edgeed from non-existent devices (the ones to the "past" pins in the first block of edges).  When generated against the MPI HW model with bucket filling and one device per thread, these phantom edges are associated with the first device, which gains four additional edges in the edge definition in the generated vars_1_0.cpp

.

Expected output:
```
outEdge_t Thread_0_Device_N0_21_OutPin_out_Tgts[2] = {{PNULL,2,0,0},{PNULL,5,0,1}};
```

Actual output:
```
outEdge_t Thread_0_Device_N0_21_OutPin_out_Tgts[6] = {{PNULL,1,0,1},{PNULL,2,0,0},{PNULL,2,0,1},{PNULL,3,0,1},{PNULL,4,0,1},{PNULL,5,0,1}};
```

heliosfa, 2021-05-27:
The relevant snippet of what is happening:

```
	<DeviceInstances>
		<DevI id="N0_21" type="node" P="0,21,0.21,90.90909,0,0.09999993"/>
		<DevI id="N1_21" type="node" P="1,21,0.21,0.00199996,0.01,0"/>
		<DevI id="N2_21" type="node" P="2,21,0.21,0.00199996,0.01,0"/>
		<DevI id="N3_21" type="node" P="3,21,0.21,0.00099999,0.01,0"/>
		<DevI id="N0_22" type="node" P="0,22,0.22,90.90909,0,0.2"/>
		<DevI id="N1_22" type="node" P="1,22,0.22,0.00199996,0.01,0"/>
		<DevI id="N2_22" type="node" P="2,22,0.22,0.00199996,0.01,0"/>
		<DevI id="N3_22" type="node" P="3,22,0.22,0.00099999,0.01,0"/>
		<DevI id="N0_23" type="node" P="0,23,0.23,90.90909,0,0.2999999"/>
		<DevI id="N1_23" type="node" P="1,23,0.23,0.00199996,0.01,0"/>
		<DevI id="N2_23" type="node" P="2,23,0.23,0.00199996,0.01,0"/>
		<DevI id="N3_23" type="node" P="3,23,0.23,0.00099999,0.01,0"/>
	</DeviceInstances>
	<EdgeInstances>
		<EdgeI path="N0_21:in-N1_21:out" P="0,0.001"/>
		<EdgeI path="N0_21:past-N0_20:out"/>
		<EdgeI path="N1_21:in-N0_21:out" P="0,0.001"/>
		<EdgeI path="N1_21:in-N2_21:out" P="0,0.001"/>
		<EdgeI path="N1_21:past-N1_20:out"/>
		<EdgeI path="N2_21:in-N1_21:out" P="0,0.001"/>
		<EdgeI path="N2_21:in-N3_21:out" P="0,0.001"/>
		<EdgeI path="N2_21:past-N2_20:out"/>
		<EdgeI path="N3_21:in-N2_21:out" P="0,0.001"/>
		<EdgeI path="N3_21:past-N3_20:out"/>

		<EdgeI path="N0_22:in-N1_22:out" P="0,0.001"/>
		<EdgeI path="N0_22:past-N0_21:out"/>
		<EdgeI path="N1_22:in-N0_22:out" P="0,0.001"/>
		<EdgeI path="N1_22:in-N2_22:out" P="0,0.001"/>
		<EdgeI path="N1_22:past-N1_21:out"/>
		<EdgeI path="N2_22:in-N1_22:out" P="0,0.001"/>
		<EdgeI path="N2_22:in-N3_22:out" P="0,0.001"/>
		<EdgeI path="N2_22:past-N2_21:out"/>
		<EdgeI path="N3_22:in-N2_22:out" P="0,0.001"/>
		<EdgeI path="N3_22:past-N3_21:out"/>

		<EdgeI path="N0_23:in-N1_23:out" P="0,0.001"/>
		<EdgeI path="N0_23:past-N0_22:out"/>
		<EdgeI path="N1_23:in-N0_23:out" P="0,0.001"/>
		<EdgeI path="N1_23:in-N2_23:out" P="0,0.001"/>
		<EdgeI path="N1_23:past-N1_22:out"/>
		<EdgeI path="N2_23:in-N1_23:out" P="0,0.001"/>
		<EdgeI path="N2_23:in-N3_23:out" P="0,0.001"/>
		<EdgeI path="N2_23:past-N2_22:out"/>
		<EdgeI path="N3_23:in-N2_23:out" P="0,0.001"/>
		<EdgeI path="N3_23:past-N3_22:out"/>
	</EdgeInstances>
```

The second edge instance references a non-existent device and the integrity checker doesn't flag it.

To debug: dump the data structure and check the def/ref flags.

DrongoTheDog, 2021-06-08:
> *CAUTION:* This e-mail originated outside the University of Southampton.
>
> Assigned #175
> <https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FPOETSII%2FOrchestrator%2Fissues%2F175&data=04%7C01%7Cadb%40soton.ac.uk%7Ca3483ea04a064a5f192508d92a9646ec%7C4a5378f929f44d3ebe89669d03ada9d8%7C0%7C0%7C637587647190305589%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=H5K2s8rzN3GyCelqNxqht2KAilcP%2BImlibmKM5e59oQ%3D&reserved=0>
> to @DrongoTheDog
> <https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FDrongoTheDog&data=04%7C01%7Cadb%40soton.ac.uk%7Ca3483ea04a064a5f192508d92a9646ec%7C4a5378f929f44d3ebe89669d03ada9d8%7C0%7C0%7C637587647190315608%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=zNd9HasZv6VEf1y5EawqJvwbN4KDO1PAnfS3%2FFCpKG0%3D&reserved=0>.

Coo.

PR #228 (2021-06-15): Complain about edges using non-existent devices.
Resolves #175. Changes from ADB, sanitised by yours truly.

Tested using the example raised in the issue.
closed 2021-06-15 by mvousden