POETSII/Orchestrator #186: orchestrator_dependences_7 does not include Qt (low priority)

m8pple, 2021-05-15:
Trying to compile using Orchestrator development branch (6669d06a9d2a7c4e6b732d8a719a0e6b69f2226d), and latest
orchestrator_dependencies release (orchestrator_dependencies_7), as suggested by the user guide.
This is on a local Ubuntu 18.04 instance, not on a Cambridge machine.

Problem:

A few files still `#include <QtCore>`, and so in `Makefile.dependencies` there is:
```
QT_DIR = $(ORCHESTRATOR_DEPENDENCIES_DIR)/Qt5.6.3/5.6.3/gcc_64
```
However, this is not in the tarball of dependencies. I assume this is part of a drive to remove Qt.

Hacky fix:

Fixed for me by just hacking `Makefile.dependencies`:
```
QT_INC_DIR = /usr/include/x86_64-linux-gnu/qt5
```
which is where Ubuntu puts the qt5 headers.

m8pple, 2021-05-16:
It seems that this is fixed by compiling the `1.0.0-alpha` branch, which works correctly with orchestrator_dependncies_7.

In both the `master` and `1.0.0-alpha` branches of `orchestrator-documentation` it talks about cloning the `development` branch:

> Obtain the sources: Clone the Orchestrator Git repository, at https://github.com/poetsii/Orchestrator, and check out the development branch.

Could `1.0.0-alpha` be merged into `development`?

heliosfa, 2021-05-16:
We are in a little bit of flux at the moment with docs and repos - development is still the old v2 Orchestrator while 1.0.0-alpha is where we were doing testing. 

A lot of improvements and fixes have been made on a feature branch, which has an open PR into alpha. There is then going to be a PR from alpha into development as everything has been pretty well tested by us, a 3rd year student and JRB.

The docs still discuss development as we were not expecting the merges, etc. To drag on so long and we only had a couple of people using the latest version.

As to the issue, the “mew” version of the Orchestrator no longer relies on QT. If you want to use an older version of the orchestrator, you will require an earlier version of the Orchestrator dependencies.

m8pple, 2021-05-16:
I realised while trying to switch between `development` and `1.0.0-alpha` that the orchestrator_dependencies versions
have different folder names. So the build instructions could say something like:

> To install orchestrator dependencies:
> - Look for the line `ORCHESTRATOR_DEPENDENCIES_DIR ?= /local/orchestrator-common/orchestrator_dependencies_N`
> - Download release N of orchestrator-dependencies
> - Untar the release file to a directory `SOMEPATH` 
> - Check that the file `SOMEPATH/orchestrator-common/orchestrator_dependencies_N` exists
> - Set the environment variable `ORCHESTRATOR_DEPENDENCIES_DIR = SOMEPATH/orchestrator_dependencies_N`