POETSII/Orchestrator #192: load /app can't find files in dir set by path /apps

m8pple, 2021-05-16:
This is in "1.0.0-alpha", commit 6c6daa6f29eabebed4516bad25540a51ffde1504.

I've got the orchestrator examples repo cloned to `/mnt/e/dt10_all/POETS/Orchestrator_examples`:
```
dt10@joxer:/mnt/e/dt10_all/POETS/Orchestrator$ ls /mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen
plate_32x32.xml  plate_33x33.xml  plate_3x3.xml
```

If I then try to set the search path using `path /apps` and then load `plate_3x3.xml` it can't find the file:
```
dt10@joxer:/mnt/e/dt10_all/POETS/Orchestrator$ rlwrap -r -H .orchestrate-cmd-history.txt ./orchestrate.sh
[WARN] Launcher: Not running on a POETS box, and found no motherships running on alternative machines, so we're not spawning any mothership processes.
POETS> 11:38:35.03:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_05_16T11_38_35p0.plog'.
POETS> 11:38:35.03: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS>path /apps = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen"
POETS> 11:38:42.37:  23(I) path /apps = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen"
POETS> 11:38:42.37:  20(I) The microlog for the command 'path /apps = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen"' will be written to '../Output/Microlog/Microlog_2021_05_16T11_38_42p0.plog'.
POETS>load /app = "plate_3x3.xml"
POETS> 11:38:54.81:  23(I) load /app = "plate_3x3.xml"
POETS> 11:38:54.81:  20(I) The microlog for the command 'load /app = "plate_3x3.xml"' will be written to '../Output/Microlog/Microlog_2021_05_16T11_38_54p0.plog'.
POETS> 11:38:54.81: 112(E) Cannot find file plate_3x3.xml
POETS>
```

However, if I give the absolute path then it can find the file:
```
dt10@joxer:/mnt/e/dt10_all/POETS/Orchestrator$ rlwrap -r -H .orchestrate-cmd-history.txt ./orchestrate.sh
[WARN] Launcher: Not running on a POETS box, and found no motherships running on alternative machines, so we're not spawning any mothership processes.
POETS> 11:39:44.04:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_05_16T11_39_44p0.plog'.
POETS> 11:39:44.04: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS>load /app = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen/plate_3x3.xml"
POETS> 11:40:10.18:  23(I) load /app = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen/plate_3x3.xml"
POETS> 11:40:10.18:  20(I) The microlog for the command 'load /app = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen/plate_3x3.xml"' will be written to '../Output/Microlog/Microlog_2021_05_16T11_40_10p0.plog'.
POETS> 11:40:10.18: 235(I) Application file /mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen/plate_3x3.xml loading...
POETS> 11:40:10.18: 201(E) ... failed validation with 40 errors in 26161 msecs
POETS>
```
(No idea why it is failing to validate, but it can at least find the file).

heliosfa, 2021-05-16:
Does it work if you change the load command to:

`load /app = +"plate_3x3.xml"`

(Note the plus before the file name). The syntax of the commands has changed to be that the plus means append to the set path, no plus is absolute.

m8pple, 2021-05-16:
Ah, I had mis-read the document about `+`. So I had interpreted it as "relative to the current
directory", and not noticed the "configured directory". I thought '+' was something new,
rather than providing the old behaviour.

I actually tried using + before and it also got `Cannot find file`:
```
dt10@joxer:/mnt/e/dt10_all/POETS/Orchestrator$ rlwrap -r -H .orchestrate-cmd-history.txt ./orchestrate.sh
[WARN] Launcher: Not running on a POETS box, and found no motherships running on alternative machines, so we're not spawning any mothership processes.
POETS> 12:05:22.06:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_05_16T12_05_22p0.plog'.
POETS> 12:05:22.06: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS>path /apps = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen"
POETS> 12:05:29.12:  23(I) path /apps = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen"
POETS> 12:05:29.12:  20(I) The microlog for the command 'path /apps = "/mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregen"' will be written to '../Output/Microlog/Microlog_2021_05_16T12_05_29p0.plog'.
POETS>load /app = +"plate_3x3.xml"
POETS> 12:05:36.64:  23(I) load /app = +"plate_3x3.xml"
POETS> 12:05:36.64:  20(I) The microlog for the command 'load /app = +"plate_3x3.xml"' will be written to '../Output/Microlog/Microlog_2021_05_16T12_05_36p0.plog'.
POETS> 12:05:36.64: 112(E) Cannot find file /mnt/e/dt10_all/POETS/Orchestrator_examples/plate_heat/pregenplate_3x3.xml
```

However, I didn't notice that it is because of a missing trailing slash when configuring the directory.

Suggestions
-------------

The documentation does describe things correctly, and I think without memories of the old
approach I would have read it more carefully and avoided the problem.

To make it more user-friendly could I suggest:

1. Change the discussion in `orchestrator-documentation` from:

     > The + operator informs the Orchestrator to look in the configured directory
     > (the root directory of the Orchestrator in the default configuration) for the application file.

   to:

    > The + operator informs the Orchestrator to look for the application file in the directory
     > configured using `path /apps`; by default this is the Orchestrator's root directory.

2. Make the default application path the current working directory when `orchestrate.sh`
    was launched. Assuming most people will be users of the orchestrator, they will be
    calling it as `/usr/local/Orchestrator/orchestrate` or something, and probably don't have
    the ability to write files in the orchestrator root directory. I'm currently trying to use the orchestrator,
    so my working directory is where the application files I'm developing are. (Also, there are
    no xml files anywhere in the orchestrator root folder).

3. Allow configured directories to be set with or without trailing slashes. It is explicitly defined as a
   directory rather than a prefix, so it should be possible to add a slash if it doesn't already
   exist.

mvousden, 2021-05-25:
Going through suggestions:

> 1.

I disagree with this phrasing, but I think we can come to a compromise... a PR is inbound.

> 2.

It already is.

> 3.

Good one.

mvousden, 2021-05-25:
Reading #193, I now understand that, by your second point, you mean people might call `orchestrator.sh` from a different directory. One thing at a time, I reckon.

closed 2021-05-25 by mvousden