POETSII/Orchestrator #196: Inconsistent information on how to run a batch script

m8pple, 2021-05-16:
In `usage.pdf` we have:
```
./orchestrate.sh -f /absolute/path/to/batch/script
```
then in the usage for `development` there is:
```
/b = FILE: Path to a batch script to run on the root process on startup.
```

Trying different variants with Orchestrator/development (), it seems that the `-b` one is still valid:
```
dt10@ayres:~/Orchestrator$ ./orchestrate.sh /f = test.script 
^C
dt10@ayres:~/Orchestrator$ ./orchestrate.sh -f = test.script 
^C
dt10@ayres:~/Orchestrator$ ./orchestrate.sh -f test.script 
+------------------------+
UIF default error handler: id 1

(Line,Col)
(   1,  6) /      <- Token (Sy_div [/                  ]) out of sequence
(   1,  4) task
(   0,  0) 
(   0,  0) 
(   0,  0) 
<snip>
...
dt10@ayres:~/Orchestrator$ ./orchestrate.sh /b = test.script 
^C
dt10@ayres:~/Orchestrator$ ./orchestrate.sh -b = test.script 
^C
dt10@ayres:~/Orchestrator$ ./orchestrate.sh -b test.script 
Attach debugger to Root process 0 (0).....

POETS>Cli dump+++++++++++++++++++++++++++++++++++++++++
Problem : FALSE
Orig = ||||
Co = ||*||
Cli dump-----------------------------------------
 13:39:00.02: 102(I) Task graph default file path is || ||
POETS> 13:39:00.02: 103(I) New path is ||/home/dt10/Orchestrator_examples/plate_heat/pregen/||
POETS>
Orchestrator has 1 tasks loaded:

    |Task       |Supervisor |Linked   |Devices  |Channels |Declare    |PoL? |PoL type   |Parameters    
    +-----------+-----------+---------+---------+---------+-----------+-----+-----------+------------+----....
  0 | plate_3x3 |plate_3x3_supervisorNode_inst |      no |      10 |      29 |plate_heat |User |           |/home/dt10/Orchestrator_examples/plate_heat/pregen/plate_3x3.xml  |
    +-----------+-----------+---------+---------+---------+-----------+-----+-----------+------------+----....
Default display filepath ||/home/dt10/Orchestrator_examples/plate_heat/pregen/||

 13:39:00.02: 138(I) Creating topology ||Simple [1 box]||
POETS> 13:39:00.02: 209(I) Attempting to place task 'plate_3x3' using the 'link' method...
POETS> 13:39:00.02: 202(I) Task 'plate_3x3' placed successfully.
POETS>cp: cannot stat '/home/dt10/Orchestrator_examples/plate_heat/pregen/../Supervisor.*': No such file or directory
 13:39:00.02: 801(D) P_builder::Add(name=plate_3x3,file=/home/dt10/Orchestrator_examples/plate_heat/pregen/plate_3x3.xml)
POETS> 13:39:00.02: 807(E) Unable to copy files to directory /home/dt10/Orchestrator_examples/plate_heat/pregen/plate_3x3Generated. Check directory exists and has access rights. Success
POETS>
POETS>
```

So there seems to be disconnect between the documentation and `/C = ARG" vs `-C ARG` styles,
and that the `/C = ARG` style doesn't work on the command-line (?) 

In the `1.0.0-alpha` documentation we have the example:
```
./orchestrate.sh -b /absolute/path/to/batch/script
```
then in the usage for `1.0.0-alpha` there is:
```
/b = FILE: Path to a batch script to run on the root process on startup. 
```
So I think the same thing might still persist. Does only `-b work`?

mvousden, 2021-05-25:
> it seems that the -b one is still valid

The appropriate flag is `-b`, which (as far as I can tell), is stated consistently across all of `1.0.0-alpha`. For reference, it was changed from `-f` in development, to support loading of files for other purposes.

> So there seems to be disconnect between the documentation and `/C = ARG` vs `-C ARG` styles, and that the `/C = ARG` style doesn't work on the command-line (?)

`/C = ARG` is incorrect. The syntax is either `/C=ARG`, or `-C ARG` (given `C` is an appropriate switch/flag).

m8pple, 2021-05-26:
Closing as it seems that 1.0.0-alpha switch resolves it.

closed 2021-05-26 by m8pple