POETSII/Orchestrator #202: What is `OrchestratorMessages.ocfg` all about? (low priority)

m8pple, 2021-05-16:
This is more just wondering, but I'm not too sure what the purpose of `OrchestratorMessages.ocfg` is.

I get that it defines printf-style format strings, and that it is used in calls to `OrchBase::Post`, but it
seems an odd level of indirection - it separates the posting call-site from the message definition,
which feels a little fragile. So if I see:
```
par->Post(235,fn);
```
then I have to go look for:
```
235(I) : "Application file %s loading..."
```
to work out what it means and check it is in sync.

Usually something like this I'd expect in an i18n resource file, but I don't
think that is what is going on here. If so, you'd expect it to be at a path
like `Config/en-GB/OrchestratorMessages.ocfg`, and then for someone
to populate the other country codes. 

They kind of feel like HTTP status codes, but are much more specific rather than
capturing the general type of error + a more specific error message.

Anyway... doesn't really matter, I was just wondering.

mvousden, 2021-05-25:
You understand entirely. I make no further comment ;p

m8pple, 2021-05-26:
Ok. Given this doesn't really matter (while odd, it is quite understandable), I'll just close this to reduce issue noise.

closed 2021-05-26 by m8pple