POETSII/Orchestrator #203: Inconsistent behaviour of command outputs for success or failure.

m8pple, 2021-05-16:
(This is based on `1.0.0-alpha` (6669d06a9d2a7c4e6b732d8a719a0e6b69f2226d), but may
also apply to `development`).

(This is related to working around #197 and #198, and trying to do automated testing
with the orchestrator.)

When trying to develop scripts that will test whether a given orchestrator passed or
failed, it is difficult to work out whether a common succeeded or failed in a uniform
way.

For example, the command `load /app` will return either `201(E)` or `65(I)`, so we
can watch for those two results and check whether it completed.

However, other commands do not print anything out. For example,  `tlink /app = *`
prints nothing if it succeeds (or maybe it fails - not sure). I guess you can look in the
microlog?

When trying to adapt the interactive interface for batch testing it would help to
have a really un-ambiguous result for each command: did it succeed, or did it fail?

Ideally there would be something that was consistent for all commands, but at least
having two distinct outputs/codes in `OrchestratorMessages.ocfg` would make it
much easier to script.

m8pple, 2021-05-16:
(Though looking for "unintelligible" is a great detector of certain classes of failure).

heliosfa, 2021-05-27:
We will be more consistent on making sure that *every* commands produces a logged output with the correct I/W/E/etc. flag.

m8pple, 2022-07-12:
Closing. Idealistic, and not really critical for anything.

closed 2022-07-12 by m8pple