POETSII/Orchestrator #204: Orchestrator 1.0.0-alpha expects message decl. as direct descendent of MessageType

m8pple, 2021-05-17:
In the v4 structure initially proposed by UoS the MessageType element has a sub-element of type Message, and then
the C-style decl is a text child within that. My vague memory was the rationale that we might want to put another element
under MessageType in the future, which is difficult if the c-style decl is a direct text child.

So in the v4 structure as agreed, and in all the example v4 XML in PIP20 and the 2019 benchmarks, the element looks
like:
```
<MessageType>
   <Message>
    float x;
    float y;
   </Message>
</MessageType>
```
However, the 1.0.0-alpha parser is expecting something of the form:
```
<MessageType>
    float x;
    float y;
</MessageType>
```

This is a breaking change for all existing v4 applications, parsers, and generators. It is less benign than
#199, as it is putting an element at a different level.

If the Orchestrator really needs this change to the xml grammar then we should bump the
Orchestrator version to v5, as it is a breaking-change at a quite a deep level.

heliosfa, 2021-05-27:
We are happy to revert to the spec. 

This will involve a change to the grammar and `DS_XML`. Plan is to "ignore" the extra level internally. Docs and examples update will likely take longer.

Suggestion is to add to the TODO list for "if EPIC"

m8pple, 2021-06-23:
This is a blocking problem, as existing v4 XML can't be parsed without it. There is a proposed
fix to make it comply with the spec in #264

heliosfa, 2021-07-02:
resolved in #264

closed 2021-07-02 by heliosfa