POETSII/Orchestrator #206: 1.0.0-alpha composer compilation error : extra '=' sign

m8pple, 2021-05-17:
When compiling a v4 example, there seems to be an extra '=' sign in the generated code, so it won't compile.

```
POETS>load /app = "/mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L3-compilation/one-dev-two-ports.xml"
Msg: mt1 = '// Line 8
// CFrag /mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L3-compilation/one-dev-two-ports.xml.Graphs.GraphType.MessageTypes.MessageType.Message empty
'
POETS> 01:45:27.04:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_05_17T01_45_27p0.plog'.
POETS> 01:45:27.04: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS> 01:45:27.04:  23(I) load /app = "/mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L3-compilation/one-dev-two-ports.xml"
POETS> 01:45:27.04:  20(I) The microlog for the command 'load /app = "/mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L3-compilation/one-dev-two-ports.xml"' will be written to '../Output/Microlog/Microlog_2021_05_17T01_45_27p1.plog'.
POETS> 01:45:27.04: 235(I) Application file /mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L3-compilation/one-dev-two-ports.xml loading...
POETS> 01:45:27.04:  65(I) Application file /mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L3-compilation/one-dev-two-ports.xml loaded in 46 ms.
POETS>tlink /app = *
place /tfill = *
POETS>POETS> 01:45:27.06:  23(I) tlink /app = *
POETS> 01:45:27.06:  20(I) The microlog for the command 'tlink /app = *' will be written to '../Output/Microlog/Microlog_2021_05_17T01_45_27p2.plog'.
POETS> 01:45:27.06:  23(I) place /tfill = *
POETS> 01:45:27.06:  20(I) The microlog for the command 'place /tfill = *' will be written to '../Output/Microlog/Microlog_2021_05_17T01_45_27p3.plog'.
POETS> 01:45:27.06: 309(I) Attempting to place graph instance 'ondev_inst' using the 'tfil' method...
POETS> 01:45:27.06: 302(I) Graph instance 'ondev_inst' placed successfully.
POETS>compose /app = *
Emit Msg __default
Emit Msg mt1
In file included from /mnt/e/dt10_all/POETS/Orchestrator/Source/Common/poets_pkt.h:7,
                 from /mnt/e/dt10_all/POETS/Orchestrator/Source/Softswitch/inc/softswitch.h:4,
                 from /mnt/e/dt10_all/POETS/Orchestrator/Source/Softswitch/inc/softswitch_common.h:4,
                 from /mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/ondev_inst__ondev_inst/Generated/inc/vars_0.h:13,
                 from /mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/ondev_inst__ondev_inst/Generated/inc/handlers_0.h:12,
                 from /mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/ondev_inst__ondev_inst/Generated/src/handlers_0.cpp:10:
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/ondev_inst__ondev_inst/Generated/src/handlers_0.cpp: In function 'uint32_t devtyp_dt1_OutPin_o1_Send_handler(const void*, void*, void*)':
/mnt/e/dt10_all/POETS/Orchestrator/Generics/OSFixes.hpp:176:30: error: expected primary-expression before '__attribute__'
 #define OS_ATTRIBUTE_UNUSED  __attribute__((unused))
                              ^~~~~~~~~~~~~
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/ondev_inst__ondev_inst/Generated/src/handlers_0.cpp:133:31: note: in expansion of macro 'OS_ATTRIBUTE_UNUSED'
    pkt_mt1_pyld_t* message =  OS_ATTRIBUTE_UNUSED= static_cast<pkt_mt1_pyld_t*>(pkt);
                               ^~~~~~~~~~~~~~~~~~~
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/ondev_inst__ondev_inst/Generated/src/handlers_0.cpp:133:20: warning: unused variable 'message' [-Wunused-variable]
    pkt_mt1_pyld_t* message =  OS_ATTRIBUTE_UNUSED= static_cast<pkt_mt1_pyld_t*>(pkt);
                    ^~~~~~~
make: *** [handlers_0.o] Error 1
make: *** Waiting for unfinished jobs....
POETS> 01:45:30.07:  23(I) compose /app = *
POETS> 01:45:30.07:  20(I) The microlog for the command 'compose /app = *' will be written to '../Output/Microlog/Microlog_2021_05_17T01_45_27p4.plog'.
POETS> 01:45:30.07: 803(I) Composing graph instance 'ondev_inst'...
POETS>Timeout or unexpected outcome while composing app.
```

heliosfa, 2021-05-17:
This issue is addressed in #172 which is pending review and merge into 1.0.0-alpha.

m8pple, 2021-05-17:
This is possible fixed by d1e684a48b8a10db15f299ff4be8fb1fa8a24299.

closed 2021-05-25 by heliosfa