POETSII/Orchestrator #207: 1.0.0-alpha composer error : extra Thread_ or space, stops generated code compiling

m8pple, 2021-05-17:
(#206 needs to be fixed before hitting this error.)

When compiling with v4 there is a space in the generated code which won't compile:
```
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/gi__gi/Generated/src/vars_0_0.cpp:21:37: error: expected initializer before 'Thread_0_Device_di1_Pin_i1_InEdgeProps'
 devtyp_dt1_InPin_i1_props_t Thread_ Thread_0_Device_di1_Pin_i1_InEdgeProps[1] = {{}};
```
I think it is something emitting an extra space.

Full log:
```
[WARN] Launcher: Not running on a POETS box, and found no motherships running on alternative machines, so we're not spawning any mothership processes.
POETS>load /app = "/mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L4-run-time/example-dt10.xml"
POETS> 01:51:29.04:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_05_17T01_51_29p0.plog'.
POETS> 01:51:29.04: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS> 01:51:29.04:  23(I) load /app = "/mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L4-run-time/example-dt10.xml"
POETS> 01:51:29.04:  20(I) The microlog for the command 'load /app = "/mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L4-run-time/example-dt10.xml"' will be written to '../Output/Microlog/Microlog_2021_05_17T01_51_29p1.plog'.
POETS> 01:51:29.04: 235(I) Application file /mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L4-run-time/example-dt10.xml loading...
POETS> 01:51:29.06:  65(I) Application file /mnt/e/dt10_all/POETS/poets_improvement_proposals/proposed/PIP-0020/xml/ic/tests/valid/L4-run-time/example-dt10.xml loaded in 46 ms.
POETS>tlink /app = *
place /tfill = *
POETS>POETS> 01:51:29.06:  23(I) tlink /app = *
POETS> 01:51:29.06:  20(I) The microlog for the command 'tlink /app = *' will be written to '../Output/Microlog/Microlog_2021_05_17T01_51_29p2.plog'.
POETS> 01:51:29.06:  23(I) place /tfill = *
POETS> 01:51:29.06:  20(I) The microlog for the command 'place /tfill = *' will be written to '../Output/Microlog/Microlog_2021_05_17T01_51_29p3.plog'.
POETS> 01:51:29.06: 309(I) Attempting to place graph instance 'gi' using the 'tfil' method...
POETS> 01:51:29.06: 302(I) Graph instance 'gi' placed successfully.
POETS>compose /app = *
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/gi__gi/Generated/src/vars_0_0.cpp:21:37: error: expected initializer before 'Thread_0_Device_di1_Pin_i1_InEdgeProps'
 devtyp_dt1_InPin_i1_props_t Thread_ Thread_0_Device_di1_Pin_i1_InEdgeProps[1] = {{}};
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/gi__gi/Generated/src/vars_0_0.cpp:25:37: error: expected initializer before 'Thread_0_Device_di1_Pin_i2_InEdgeProps'
 devtyp_dt1_InPin_i2_props_t Thread_ Thread_0_Device_di1_Pin_i2_InEdgeProps[1] = {{{4}}};
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/gi__gi/Generated/src/vars_0_0.cpp:35:37: error: expected initializer before 'Thread_0_Device_di2_Pin_i2_InEdgeProps'
 devtyp_dt1_InPin_i2_props_t Thread_ Thread_0_Device_di2_Pin_i2_InEdgeProps[1] = {{{4}}};
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/gi__gi/Generated/src/vars_0_0.cpp:37:82: error: braces around scalar initializer for type 'float'
 devtyp_dt1_InPin_i2_state_t Thread_0_Device_di2_Pin_i2_InEdgeStates[1] = {{{0.3}}};
                                                                                  ^
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/gi__gi/Generated/src/vars_0_0.cpp:50:82: error: braces around scalar initializer for type 'uint8_t' {aka 'unsigned char'}
 devtyp_dt1_props_t Thread_0_DeviceProperties[3] = {{},{{1,{2,3,4}}},{{1,{2,3,4}}}};
                                                                                  ^
/mnt/e/dt10_all/POETS/Orchestrator/Output/Composer/gi__gi/Generated/src/vars_0_0.cpp:52:72: error: braces around scalar initializer for type 'float'
 devtyp_dt1_state_t Thread_0_DeviceState[3] = {{},{},{{0.1, 2, {1,2,3}}}};
                                                                        ^
```

heliosfa, 2021-05-17:
This issue is addressed in #172 which is pending review and merge into 1.0.0-alpha.

m8pple, 2021-05-17:
Ok, I'll give up on these changes. There look like a lot of fixes backed up in that PR.

closed 2021-05-25 by heliosfa