POETSII/Orchestrator #221: XML parser rejects re-declaration of xmlns

m8pple, 2021-05-27:
The XML parser treats `xmlns` as a normal attribute, so if any element other than
`Graphs` declares a namespace (even to re-declare the same namespace), then it
is rejected. Re-declaration is fairly common when pasting together fragments of
XML, and should be invisible outside the parser. For example, this occurs in one of
the example v4 files sent to Andrew to help him ~reinvent the wheel~ write an
XML parser from scratch:

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/xml/ic/apps/betweeness_centrality_16_16_20_20_v4.xml#L1-L3

I think the XML parser is currently unaware of namespaces (e.g. #200), so this is
not something easily fixed properly.

Suggested workaround is to add an optional xmlns attribute to V4Grammar3.ocfg on the GraphType and GraphInstance
elements, as these are the most likely elements to re-declare their namespace.

m8pple, 2021-05-27:
Workaround in this commit: 74e27478dee3e7031cda1c608e8abb1080a83756

m8pple, 2021-06-23:
I'm not sure what you mean by changing the grammar. Re-declaring xmlns is a fundamental
property of XML. The patch I proposed changes it in two places in the ocfg grammar.

So I've rebased the proposed patch from last month into #264 , as it is still a blocking issue.

heliosfa, 2021-07-02:
resolved in #264

closed 2021-07-02 by heliosfa