POETSII/Orchestrator #222: Changed spelling of Metadata plus extended locations

m8pple, 2021-05-27:
In the v4 spec the element name is `Metadata`, and that was used in the example xml given
to Andrew. But in the orchestrator it is given as `MetaData`. Depending on how much xml
is floating around with the `MetaData` form, we could just change the spec to say that
either form is correct?

I also notice the grammar allows the meta-data to be attached to lots of different
elements, which was true of v3 and earlier, but in the v4 discussion it was required
that it _only_ be attached to GraphType or GraphInstance, and nowhere else.

I'm happy to open it up, as it is a non-breaking change, and I didn't really get the
reasoning for restricting it in the first place.

m8pple, 2021-05-27:
An actual bug is that in the v4 grammar the elements have to be in a strict order,
and it was specified the `MetaData` comes before `Properties`:

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L238-L255

In the orchestrator grammar they are switched:

https://github.com/POETSII/Orchestrator/blob/b5615c33a5476c4ed1d1170cbc1abaa4201ad007/Config/V4Grammar3.ocfg#L4-L6

m8pple, 2021-05-27:
Fixed by this commit: 74e27478dee3e7031cda1c608e8abb1080a83756

heliosfa, 2021-05-27:
We are happy to revert to the specified case ("Metadata") as we don't currently do anything with it.

m8pple, 2021-06-23:
I've rebased the proposed patch from last month into #264 , as it is still a blocking issue.

heliosfa, 2021-07-02:
Resolved in #264

closed 2021-07-02 by heliosfa