POETSII/Orchestrator #230: ampersand is replaced with dollar during code generation

m8pple, 2021-06-12:
This was done on development_dt10_branch (e992e8b7f16481e06f32a4fb794a4865cef9c485), as otherwise I
cant get through parsing through to compilation. The branch has the current 1.0.0-alpha merged in.

During code generation some ampersands in handlers get changed into dollars. For example, this:
```
<ReadyToSend><![CDATA[
        *readyToSend=0;
        if( (deviceState->i < 16) && (0==(deviceState->i&1)) ){
            *readyToSend=RTS_FLAG_problem;
        }else if(deviceState->i == 16){
            *readyToSend=RTS_FLAG_finished;
        }
        handler_log(2, "test RTS : 0x%x", *readyToSend);
        ]]></ReadyToSend>
```
gets changed into:
```
uint32_t devtyp_tester_RTS_handler (const void* __GraphProps, void* __Device, uint32_t* readyToSend)
{
    const global_props_t* graphProperties OS_ATTRIBUTE_UNUSED= static_cast<const global_props_t*>(__GraphProps);
    OS_PRAGMA_UNUSED(graphProperties)
   PDeviceInstance* deviceInstance OS_ATTRIBUTE_UNUSED= static_cast<PDeviceInstance*>(__Device);
   OS_PRAGMA_UNUSED(deviceInstance)
    const devtyp_tester_props_t* deviceProperties OS_ATTRIBUTE_UNUSED= static_cast<const devtyp_tester_props_t*>(deviceInstance->properties);
    OS_PRAGMA_UNUSED(deviceProperties)
    const devtyp_tester_state_t* deviceState OS_ATTRIBUTE_UNUSED= static_cast<devtyp_tester_state_t*>(deviceInstance->state);
    OS_PRAGMA_UNUSED(deviceState)
// Line 582

        *readyToSend=0;
        if( (deviceState->i < 16) && (0==(deviceState->i$1)) ){
            *readyToSend=RTS_FLAG_problem;
        }else if(deviceState->i == 16){
            *readyToSend=RTS_FLAG_finished;
        }
        handler_log(2,"test RTS : 0x%x", *readyToSend);
        
    return *readyToSend;
}
```

The specific problem is:
```
        if( (deviceState->i < 16) && (0==(deviceState->i&1)) ){
```
versus:
```
        if( (deviceState->i < 16) && (0==(deviceState->i$1)) ){
```

As a consequence, this fails to compose:
```
dt10@joxer:~/POETS/Orchestrator$ ./orchestrate.sh 
[WARN] Launcher: Not running on a POETS box, and found no motherships running on alternative machines, so we're not spawning any mothership processes.
POETS> 11:39:51.02:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_06_12T11_39_51p0.plog'.
POETS> 11:39:51.02: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS>
POETS>load /app = "/home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/amg_poisson_8_8_v4.xml"
Msg: linear_problem = '// Line 31
float x;
float b;
'
Msg: linear_solution = '// Line 37
float x;
float r;
int32_t i;
'
Msg: jacobi_exchange = '// Line 42
float x;
uint32_t tag;
[... 73 lines cut ...]
riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o vars_96.o /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/vars_96.cpp
riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o handlers_96.o /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_96.cpp
/home/dt10/POETS/Orchestrator/Source/Softswitch/src/genld.sh 96 > link_96.ld
riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o vars_64_0.o /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/vars_64_0.cpp
riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o vars_64.o /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/vars_64.cpp
riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o handlers_64.o /home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_64.cpp
/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_96.cpp: In function 'uint32_t devtyp_tester_RTS_handler(const void*, void*, uint32_t*)':
/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_96.cpp:47:56: error: 'const devtyp_tester_state_t' {aka 'const struct tester_state_t'} has no member named 'i$1'
         if( (deviceState->i < 16) && (0==(deviceState->i$1)) ){
                                                        ^~~
/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_96.cpp: In function 'uint32_t devtyp_tester_InPin_solution_Recv_handler(const void*, void*, void*, const void*)':
/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_96.cpp:156:34: error: 'devtyp_tester_state_t' {aka 'struct tester_state_t'} has no member named 'i$1'
             assert((deviceState->i$1)==1);
                                  ^~~
/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_96.cpp: In function 'uint32_t devtyp_tester_OutPin_problem_Send_handler(const void*, void*, void*)':
/home/dt10/POETS/Orchestrator/Output/Composer/amg_poisson_8x8__amg_poisson_8x8/Generated/src/handlers_96.cpp:211:34: error: 'devtyp_tester_state_t' {aka 'struct tester_state_t'} has no member named 'i$1'
             assert((deviceState->i$1)==0);
                                  ^~~
Makefile:192: recipe for target 'handlers_96.o' failed
make: *** [handlers_96.o] Error 1
make: *** Waiting for unfinished jobs....
rm vars_64.o handlers_64.o softswitch_main.o softswitch_common.o vars_128.o vars_32.o handlers_128.o handlers_32.o link_128.ld vars_96.o vars_0.o link_32.ld handlers_0.o link_96.ld link_0.ld

Compilation failed! make_errs.txt dump ----------------------------------------
```

m8pple, 2021-06-12:
Couldn't easily see where the problem was, but changing the input from:
```
if( (deviceState->i < 16) && (0==(deviceState->i&1)) ){
```
to
```
if( (deviceState->i < 16) && (0==(deviceState->i& 1)) ){
```

So searching and replacing `&1` with `& 1` is a workaround.

mvousden, 2021-06-14:
Worth noting, `&` is ASCII 38, and `$` is ASCII 36 (decimal). Did you observe that the transformation was consistent?

mvousden, 2021-06-14:
Found it. @m8pple: I've opened a PR, but would be grateful if you could either test with your example, or send your example to me to test with.

m8pple, 2021-06-14:
It's one of the v4 reference graphs, so available here:

https://github.com/POETSII/poets_improvement_proposals/blob/master/proposed/PIP-0020/xml/ic/apps/amg_poisson_8_8_v4.xml

mvousden, 2021-06-14:
Running that example, I get a bunch of other structural errors, which appear to be related to other issues. Can you please drop the XML you ran here?

Errors, for reference:

```
........................................................................................................................
14/06/2021 14:04:37.06

Checking client file ../ring_test.xml
against grammar file ../Config/V4Grammar3.ocfg

(lin,col) refers to element closure location in client file

Structural analysis:

  1. (1864,  9) : Element ../ring_test.xml.Graphs
                  contains 0 instance(s) of attribute "appname": this (0) should be exactly 1

  2. (  32, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

  3. (  32, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 1 instance(s) of element "Message": this (1) should be exactly 0

  4. (  31, 13) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType.Message
                  contains 1 instance(s) of element "CDATA": this (1) should be exactly 0

  5. (  38, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

  6. (  38, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 1 instance(s) of element "Message": this (1) should be exactly 0

  7. (  37, 13) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType.Message
                  contains 1 instance(s) of element "CDATA": this (1) should be exactly 0

  8. (  43, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

  9. (  43, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 1 instance(s) of element "Message": this (1) should be exactly 0

 10. (  42, 13) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType.Message
                  contains 1 instance(s) of element "CDATA": this (1) should be exactly 0

 11. (  48, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 12. (  48, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 1 instance(s) of element "Message": this (1) should be exactly 0

 13. (  47, 13) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType.Message
                  contains 1 instance(s) of element "CDATA": this (1) should be exactly 0

 14. (  53, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 15. (  53, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 1 instance(s) of element "Message": this (1) should be exactly 0

 16. (  52, 13) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType.Message
                  contains 1 instance(s) of element "CDATA": this (1) should be exactly 0

 17. (  56, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 18. (  56, 20) : Element ../ring_test.xml.Graphs.GraphType.MessageTypes.MessageType
                  contains 1 instance(s) of element "Message": this (1) should be exactly 0

 19. (  71, 18) : Element ../ring_test.xml.Graphs.GraphType.DeviceTypes.DeviceType.InputPin.State
[... 72 lines cut ...]
                  contains 1 instance(s) of attribute "indexed": this (1) should be exactly 0

 44. ( 584, 25) : Element ../ring_test.xml.Graphs.GraphType.DeviceTypes.DeviceType.OnHardwareIdle
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 45. ( 585, 23) : Element ../ring_test.xml.Graphs.GraphType.DeviceTypes.DeviceType.OnDeviceIdle
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 46. ( 594, 23) : Element ../ring_test.xml.Graphs.GraphType.DeviceTypes.DeviceType.InputPin.Properties
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 47. ( 595, 18) : Element ../ring_test.xml.Graphs.GraphType.DeviceTypes.DeviceType.InputPin.State
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 48. ( 607, 25) : Element ../ring_test.xml.Graphs.GraphType.DeviceTypes.DeviceType.OnHardwareIdle
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

 49. ( 608, 23) : Element ../ring_test.xml.Graphs.GraphType.DeviceTypes.DeviceType.OnDeviceIdle
                  contains 0 instance(s) of element "CDATA": this (0) should be exactly 1

...Client file ../ring_test.xml
 exhibits 49 accumulated (syntax/structure) errors in 63 msecs

------------------------------------------------------------------------------------------------------------------------
```

m8pple, 2021-06-14:
I probably don't have a version of this file that will get far enough in stock 1.0.0-alpha, as the branch
I'm mentioned (e992e8b7f16481e06f32a4fb794a4865cef9c485) contains multiple fixes.

I manually applied the patch to my branch, then un-patched the XML to get the &1 back, and then tried. Switching
that character from $ to & and back seems to change it from broken to working to broken, so I think it has
fixed it.

PR #239 (2021-06-15): Don't transform ampersands to dollars.
Don't approve this, until @m8pple has tested it with his XML. I've tested with something I wrote myself, but I couldn't find a copy of the one he used.

Resolves #230.
closed 2021-06-15 by mvousden