POETSII/Orchestrator #231: Standard type-name aliases are not emitted

m8pple, 2021-06-12:
This was done on development_dt10_branch (e992e8b), as otherwise I
cant get through parsing through to compilation. The branch has the current 1.0.0-alpha merged in.

According to the v4 spec, there should be a type called `"{graphTypeId}_{deviceTypeId}_state_t"` which
allows one to refer to the type-name for a given state:

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L402-L412

It looks like there is an internal type called `devtyp_{deviceTypeId}_state_t`, but no alias for the implementation independent name.

In cases where the shared code uses the implementation independent name, it fails to compile.

For example:
```
dt10@joxer:~/POETS/Orchestrator$ ./orchestrate.sh 
[WARN] Launcher: Not running on a POETS box, and found no motherships running on alternative machines, so we're not spawning any mothership processes.
POETS> 12:13:14.03:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_06_12T12_13_13p0.plog'.
POETS> 12:13:14.03: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS>load /app = "/home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml"
Msg: share = '// Line 42
uint32_t source;
uint32_t distances[8];
'
Msg: stats = '// Line 48
uint64_t sum_sum_distance;
uint32_t vertex_count;
uint32_t max_distances[8];
'
POETS> 12:13:17.07:  23(I) load /app = "/home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml"
POETS> 12:13:17.07:  20(I) The microlog for the command 'load /app = "/home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml"' will be written to '../Output/Microlog/Microlog_2021_06_12T12_13_16p0.plog'.
POETS> 12:13:17.07: 235(I) Application file /home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml loading...
POETS> 12:13:17.07:  65(I) Application file /home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml loaded in 3074 ms.
POETS>tlink /app = *
POETS> 12:13:20.20:  23(I) tlink /app = *
POETS> 12:13:20.20:  20(I) The microlog for the command 'tlink /app = *' will be written to '../Output/Microlog/Microlog_2021_06_12T12_13_20p0.plog'.
POETS>place /tfill = *
POETS> 12:13:23.19:  23(I) place /tfill = *
POETS> 12:13:23.19:  20(I) The microlog for the command 'place /tfill = *' will be written to '../Output/Microlog/Microlog_2021_06_12T12_13_23p0.plog'.
POETS> 12:13:23.19: 309(I) Attempting to place graph instance 'apsp_150_10' using the 'tfil' method...
POETS> 12:13:23.19: 302(I) Graph instance 'apsp_150_10' placed successfully.
POETS>compose /app = *
POETS> 12:13:33.13:  23(I) compose /app = *
POETS> 12:13:33.13:  20(I) The microlog for the command 'compose /app = *' will be written to '../Output/Microlog/Microlog_2021_06_12T12_13_26p0.plog'.
POETS> 12:13:33.13: 803(I) Composing graph instance 'apsp_150_10'...
POETS> 12:13:33.13: 806(W) Graph instance 'apsp_150_10' compose failed. Check the microlog for details.
POETS>
POETS>
```

With microlog:
```
========================================================================================================================
12/06/2021 12:13:26.11 file ../Output/Microlog/Microlog_2021_06_12T12_13_26p0.plog
command [compose /app = *]
from console
========================================================================================================================

Composing apsp_150_10...
	Generating Supervisor...	Done!
	Generating code for 2 cores
	Core 0: Files... Source... 
	Core 32: Files... Source... 
RTS Buffer for thread 0 expanded from 1 to 10

	Make called with make -j$(nproc --ignore=4) all SOFTSWITCH_TRIVIAL_LOG_HANDLER=1 SOFTSWITCH_LOGLEVEL=2 > make_errs.txt 2>&1
Compilation failed! make_errs.txt dump ++++++++++++++++++++++++++++++++++++++++

riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o softswitch.o /home/dt10/POETS/Orchestrator/Source/Softswitch/src/softswitch.cpp
riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o vars_0_0.o /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/src/vars_0_0.cpp
riscv32-unknown-elf-g++ -mabi=ilp32 -march=rv32imf -static -mcmodel=medany -fvisibility=hidden -nostartfiles -pipe -fsingle-precision-constant -fno-builtin-printf -ffp-contract=off -std=c++14 -Wall -O2  -DTRIVIAL_LOG_HANDLER -DP_LOG_LEVEL=2 -I /home/dt10/POETS/Orchestrator/Tinsel/include -I /home/dt10/POETS/Orchestrator/Source/Softswitch/inc -I /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/inc -I/home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated -I /home/dt10/POETS/Orchestrator/Generics -I /home/dt10/POETS/Orchestrator/Source/Common -Wall -c -o io.o /home/dt10/POETS/Orchestrator/Tinsel/lib/io.c
[... 39 lines cut ...]
   const unsigned K_STATE=sizeof(std::declval<apsp_vec_barrier_vertex_state_t>().distances) / sizeof(std::declval<apsp_vec_barrier_vertex_state_t>().distances[0]);
                                                                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/src/handlers_0.cpp:24:114: note: suggested alternative: 'apsp_vec_barrier_stats_message_t'
   const unsigned K_STATE=sizeof(std::declval<apsp_vec_barrier_vertex_state_t>().distances) / sizeof(std::declval<apsp_vec_barrier_vertex_state_t>().distances[0]);
                                                                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                                                                  apsp_vec_barrier_stats_message_t
/home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/src/handlers_0.cpp:24:147: error: no matching function for call to 'declval<<expression error> >()'
   const unsigned K_STATE=sizeof(std::declval<apsp_vec_barrier_vertex_state_t>().distances) / sizeof(std::declval<apsp_vec_barrier_vertex_state_t>().distances[0]);
                                                                                                                                                   ^
In file included from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/bits/move.h:55,
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/bits/stl_pair.h:59,
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/bits/stl_algobase.h:64,
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/bits/char_traits.h:39,
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/string:40,
                 from /home/dt10/POETS/Orchestrator/Generics/OSFixes.hpp:3,
                 from /home/dt10/POETS/Orchestrator/Source/Common/poets_pkt.h:7,
[... cut ...]
                 from /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/inc/handlers_0.h:18,
                 from /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/src/handlers_0.cpp:16:
/mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/type_traits:2056:10: note: candidate: 'template<class _Tp> decltype (__declval<_Tp>(0)) std::declval()'
     auto declval() noexcept -> decltype(__declval<_Tp>(0))
          ^~~~~~~
```

m8pple, 2021-06-12:
This can be fixed by adding an additional type alias in `Composer::formDevTPropsDStateD`:
```
void Composer::formDevTPropsDStateD(GraphI_t *graph,  devTypStrings_t* dTypStrs)
{
    ...
    std::string graphTName=graph->pT->Name();
    ...
    if (devT->pStateD)
    {
        vars_h << "typedef struct " << devTName << "_state_t \n{\n";
        vars_h << devT->pStateD->C_src();
        vars_h << "\n} devtyp_" << devTName << "_state_t;\n\n";
 
        // New alias
        vars_h << "typedef devtyp_" << devTName << "_state_t "<<graphTName<<"_"<<devTName<<"_state_t;\n\n";
    }
```

m8pple, 2021-06-12:
The same issue appears with ""{graphTypeId}_{deviceTypeId}_properties_t", and can be fixed in the same way.

heliosfa, 2021-06-14:
This one is my bad. I missed making the change from `devtyp_` in our v2 Softswitch to the appname one. Will look over your commits and fix.

PR #276 (2021-12-16): Fix availability and naming of properties and state structs for devices
This PR:
* Renames properties and state structures from the incorrect `devtyp_{DEVICETYPE}_...` format to `{GRAPHTYPE}_{DEVICETYPE}_...`. (fixes #231)
* Move the struct declarations to their own header, `DeviceStructs.h`, that is used by all devices and the Supervisor. (fixes #232)
* Adds `#pragma pack(push,1)` to the message stuct header. (partially addresses #240)
* Fixes a couple of issues in the `apsp_vec_barrier_150_10.xml` example that stoped it compiling.

The plate heat example has been updated, and a very simple c [... cut]
closed 2021-12-16 by heliosfa