POETSII/Orchestrator #232: Types for device properties and state only available within that devices handlers

m8pple, 2021-06-12:
This follows on from #231.

This was done on development_dt10_branch (fd0196afba08e70beafd4b87090ee15c0befb00f), as otherwise I
cant get through parsing through to compilation. The branch has the current 1.0.0-alpha merged in.

Some existing graphs refer to various global types in order to work out things like the maximum
size of things at compile-time. Also, within the global SharedCode it might refer to any of the
state or properties of devices.

At the moment it looks like only the types for the device currently being compiled are emitted, which breaks
applications that use multiple device types and refer to those types from across handlers.

For example, this is used in an APSP implementation to statically work out batch size:

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/xml/ic/apps/apsp_vec_barrier_150_10.xml#L12-L15

Current behaviour:
```
dt10@joxer:~/POETS/Orchestrator$ ./orchestrate.sh 
[WARN] Launcher: Not running on a POETS box, and found no motherships running on alternative machines, so we're not spawning any mothership processes.
POETS> 12:32:49.03:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_06_12T12_32_49p0.plog'.
POETS> 12:32:49.03: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS>load /app = "/home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml"
Msg: share = '// Line 43
uint32_t source;
uint32_t distances[8];
'
Msg: stats = '// Line 49
uint64_t sum_sum_distance;
uint32_t vertex_count;
uint32_t max_distances[8];
'
POETS> 12:32:51.05:  23(I) load /app = "/home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml"
POETS> 12:32:51.05:  20(I) The microlog for the command 'load /app = "/home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml"' will be written to '../Output/Microlog/Microlog_2021_06_12T12_32_51p0.plog'.
POETS> 12:32:51.05: 235(I) Application file /home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml loading...
POETS> 12:32:51.05:  65(I) Application file /home/dt10/POETS/Orchestrator/Tests/ReferenceXML/v4/PEP20/apps/apsp_vec_barrier_150_10.xml loaded in 2054 ms.
POETS>tlink /app = *
POETS> 12:32:54.75:  23(I) tlink /app = *
POETS> 12:32:54.75:  20(I) The microlog for the command 'tlink /app = *' will be written to '../Output/Microlog/Microlog_2021_06_12T12_32_54p0.plog'.
POETS>place /tfill = *
POETS> 12:32:56.99:  23(I) place /tfill = *
POETS> 12:32:56.99:  20(I) The microlog for the command 'place /tfill = *' will be written to '../Output/Microlog/Microlog_2021_06_12T12_32_56p0.plog'.
POETS> 12:32:56.99: 309(I) Attempting to place graph instance 'apsp_150_10' using the 'tfil' method...
POETS> 12:32:56.99: 302(I) Graph instance 'apsp_150_10' placed successfully.
POETS>compose /app = *
POETS> 12:33:06.35:  23(I) compose /app = *
POETS> 12:33:06.35:  20(I) The microlog for the command 'compose /app = *' will be written to '../Output/Microlog/Microlog_2021_06_12T12_32_58p0.plog'.
POETS> 12:33:06.35: 803(I) Composing graph instance 'apsp_150_10'...
POETS> 12:33:06.35: 806(W) Graph instance 'apsp_150_10' compose failed. Check the microlog for details.
POETS>
```

Microlog:
```
========================================================================================================================
12/06/2021 12:32:58.32 file ../Output/Microlog/Microlog_2021_06_12T12_32_58p0.plog
command [compose /app = *]
from console
========================================================================================================================

Composing apsp_150_10...
	Generating Supervisor...	Done!
	Generating code for 2 cores
	Core 0: Files... Source... 
	Core 32: Files... Source... 
RTS Buffer for thread 0 expanded from 1 to 10

	Make called with make -j$(nproc --ignore=4) all SOFTSWITCH_TRIVIAL_LOG_HANDLER=1 SOFTSWITCH_LOGLEVEL=2 > make_errs.txt 2>&1
Compilation failed! make_errs.txt dump ++++++++++++++++++++++++++++++++++++++++

[... 182 lines cut ...]
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/bits/stl_pair.h:59,
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/bits/stl_algobase.h:64,
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/bits/char_traits.h:39,
                 from /mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/string:40,
                 from /home/dt10/POETS/Orchestrator/Generics/OSFixes.hpp:3,
                 from /home/dt10/POETS/Orchestrator/Source/Common/poets_pkt.h:7,
                 from /home/dt10/POETS/Orchestrator/Source/Softswitch/inc/softswitch.h:4,
                 from /home/dt10/POETS/Orchestrator/Source/Softswitch/inc/softswitch_common.h:4,
                 from /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/inc/vars_32.h:19,
                 from /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/inc/handlers_32.h:18,
                 from /home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/src/handlers_32.cpp:16:
/mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/type_traits:2056:10: note: candidate: 'template<class _Tp> decltype (__declval<_Tp>(0)) std::declval()'
     auto declval() noexcept -> decltype(__declval<_Tp>(0))
          ^~~~~~~
/mnt/e/dt10_all/POETS/orchestrator_dependencies_7/riscv32-compile-driver/riscv32-unknown-elf/include/c++/8.2.0/type_traits:2056:10: note:   template argument deduction/substitution failed:
/home/dt10/POETS/Orchestrator/Output/Composer/apsp_150_10__apsp_150_10/Generated/src/handlers_32.cpp:30:197: error: template argument 1 is invalid
   const unsigned num_debug_2nd_round_distances = sizeof(std::declval<apsp_vec_barrier_vertex_properties_t>().debug_2nd_round_distances) / sizeof(std::declval<apsp_vec_barrier_vertex_properties_t>().debug_2nd_round_distances[0]);
                                                                                                                                                                                                     ^
Makefile:192: recipe for target 'handlers_32.o' failed
make: *** [handlers_32.o] Error 1
rm softswitch_main.o softswitch_common.o vars_32.o vars_0.o link_32.ld handlers_0.o link_0.ld

Compilation failed! make_errs.txt dump ----------------------------------------

```

mvousden, 2021-06-14:
Perhaps I'm misunderstanding, but could one just define a type in the `Graphs/GraphType/SharedCode` element, and refer to the type from here? You could then define an instance of that type in your relevant CDATA section?

heliosfa, 2021-06-14:
> Perhaps I'm misunderstanding, but could one just define a type in the Graphs/GraphType/SharedCode element, and refer to the type from here? You could then define an instance of that type in your relevant CDATA section?

You used to have to do this in our "v2" Softswitch for the local device shared code and it was a pain. 

When we (you @DrongoTheDog  and I) were discussing struct scoping, we decided that it made sense to make the device-level structs available in the device's shared code but that making them available at the global level could cause issues. Unfortunately I can't remember what the issues we identified were, but compile time springs to mind.

> Also, within the global SharedCode it might refer to any of the state or properties of devices.

Is this just for sizing? or are there other "legitimate" reasons you can see for devices (and by extension supervisors) to have access to the struct declarations for all device types?

I guess that a potential fix is to emit a general types header (rather than just a message types header) that includes all of the extra structs. However, where does this stop? Does it include all of the edge props/state structs as well?

m8pple, 2021-06-14:
Yes, you could create a custom type in `Graphs/GraphType/SharedCode`, but that type would be completely
independent of the types used for things like messages, states, and properties.

You could replicate the structure of a message type, and then make sure you kept the
two definitions in sync, but that is a recipe for problems, as experience suggests that
the widths of types get changes slightly or someone inserts another variable, but
then forget to update them.

There is also a worry that an implementation might decide to lay out the types
differently in memory - for example, some part compiled in gcc-x86 and some part
compiled in risc-v, which relies on implementation-defined padding and packing
decisions. That's why it is required that all structs are emitted with `#pragma pack(1)`,
as otherwise it is impossible to move the around portably.

(I notice that the orchestrator doesn't emit pragma pack 1. I guess this is less of
an issue while externals are not supported, but does suggest some quite interesting
edge cases between the Mothership and the devices...)

Also, if you take the approach of having two distinct types you need to either memcpy
between the two (wasting time), or type-pun them via pointers, which means the
handlers become non-synthesisable in HLS. (A HLS POETS implementation existed in the
past, and may exist again in the future).

So the well-known type-name is there to allow handlers to refer to the types
as emitted by the implementation, as it avoids a lot of those problems. Note
that there is also the `cTypeName` override, though I haven't looked into
whether the orchestrator supports that.

> You could then define an instance of that type in your relevant CDATA section?

I assume by CDATA section you mean something like a MessageType or State element?
That approach will work with an implementation that does not introspect types and
views them as strings (such as the orchestrator), but many other implementations
parse and introspect all types and values in order to discover errors, and to enable
direct binary generation (rather than getting gcc to do it).

In the v3 XML there was more support for custom type instances through typedefs,
but that was seen as too complicated to implement, so was removed in v4. So in
v4 the only things allows in the typedDataSpec and typedDataValue fields must
be standalone, with the only allowable types coming from cstdint:

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L847-L880

m8pple, 2021-06-14:
> When we (you @DrongoTheDog and I) were discussing struct scoping, we decided that it made sense to make the device-level structs available in the device's shared code but that making them available [...]

It's done in various other implementations (at least two simulators in graph_schema, poets_eco_system, and the reference
parser/implementation developed during the PIP20 discussions.

When exposed as global type names, you get `{graphTypeId}_{deviceTypeId}_properties_t`, so it shouldn't be
possible for them to conflict. Within each handler there are supposed to be aliases `DEVICE_STATE_T` and
`DEVICE_PROPERTIES_T`, but those are local to each handler (it appears unspecified whether they are in
scope for the SharedCode section of a device - I think they were intended to be, but it is a little
implicit).

> Is this just for sizing? or are there other "legitimate" reasons you can see for devices (and by extension supervisors) to have access to the struct declarations for all device types?

It's mainly todo with sizing, and for things like static assertions. It's useful to check at compile-time
that the parameters associated with one handler are still compatible with the current configuration
of another parameter.

It was more useful with the V3 support for typedefs, but a few of those use-cases have disappeared
with the simplification for V4.

In cases where you are working around the lack of expressiveness, it is still useful to be able to
static_assert that your type-punning is actually valid. For example, if a particular struct is being
serialised in a raw byte message (to get around lack of unions and typedefs), it is useful to
check that the size of the struct being sent is the same as the expected one being received.

> I guess that a potential fix is to emit a general types header (rather than just a message types header) that includes all of the extra structs. However, where does this stop? Does it include all of the edge props/state structs as well?

Yes, the spec expects named types for edge properties and state too:

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L434

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L445

PR #276 (2021-12-16): Fix availability and naming of properties and state structs for devices
This PR:
* Renames properties and state structures from the incorrect `devtyp_{DEVICETYPE}_...` format to `{GRAPHTYPE}_{DEVICETYPE}_...`. (fixes #231)
* Move the struct declarations to their own header, `DeviceStructs.h`, that is used by all devices and the Supervisor. (fixes #232)
* Adds `#pragma pack(push,1)` to the message stuct header. (partially addresses #240)
* Fixes a couple of issues in the `apsp_vec_barrier_150_10.xml` example that stoped it compiling.

The plate heat example has been updated, and a very simple c [... cut]
closed 2021-12-16 by heliosfa