POETSII/Orchestrator #235: MPI spinning consumes excessive battery and interferes with other apps

m8pple, 2021-06-14:
I'd noticed that trying to do any work with the orchestrator killed my laptop battery, to
the point that I'd mostly stopped trying. Then I accidentally left the orchestrator open,
and realised (after it had burnt half my battery), that it was the spinning threads
that were running permanently even if the orchestrator was idle.

Many users of the orchestrator are going to be developing applications and spending
most of their time getting them compiling and parsing (and eventually simulating), and
they are quite likely to be doing this on laptops. Every student we have had working
with POETS so far has worked on their laptop, and only connected to a hardware server once
it was all debugged and working in simulation.

So it would be nice if the spinner adapted if nothing is happening, e.g. some kind of
exponential backoff down to 10ms if there is repeatedly nothing there. I can see
something mentioned in the Mothership spinner, but this seems to be related to
debugging rather than efficiency.

Presumably this would be added here:

https://github.com/POETSII/Orchestrator/blob/372ccc251832256a33c59bb3c80c2199efac298a/Source/Common/CommonBase.cpp#L101-L109

m8pple, 2021-06-14:
There seems to be an API issue in that the OnIdle method cannot tell the MPISpinner code that it has idle
code to do, or that even if it did do something that it would like the opportunity to do something again.
So I guess there is no way for MPISpinner to know whether it is wasting its time or if the OnIdle method
is actually driving something.

m8pple, 2021-06-14:
Suggested fix for two threads that spin is in  dt10_idle_work_reduction

mvousden, 2021-06-14:
I like this - something I was meaning to get around to at some point, but never designed properly.