POETSII/Orchestrator #237: RTCL uses un-sychronised cross-thread comms

m8pple, 2021-06-14:
RTCL uses a set of variables in a shared struct:

https://github.com/POETSII/Orchestrator/blob/372ccc251832256a33c59bb3c80c2199efac298a/Source/RTCL/RTCL.h#L27-L35

One thread is reading from this struct :

https://github.com/POETSII/Orchestrator/blob/372ccc251832256a33c59bb3c80c2199efac298a/Source/RTCL/RTCL.cpp#L15-L25

And another thread is writing to it :

https://github.com/POETSII/Orchestrator/blob/372ccc251832256a33c59bb3c80c2199efac298a/Source/RTCL/RTCL.cpp#L140-L150

However, there is no threading synchronisation, so (assuming that those variables ever do change - I'm not sure), there
is no guarantee that any changes will ever get migrated from one thread to another. There needs to be use of
atomics or mutexes to actually ensure that hardware will always migrate, and to ensure that the compiler doesn't
make optimisation assumptions.

Probably it's working by chance at the moment, but it might stop at any point.

heliosfa, 2021-06-17:
@DrongoTheDog "Yes, there is a comms pool in the middle. What is the problem with that?"

"In general you are correct, but the reading thread is spinning so will pick up any change in future iterations."

This does away with the need for atomics/mutexes as we have the constant spinner any way (for design reasons)

m8pple, 2021-06-22:
I think you're assuming that writes from one thread will automatically appear on
another thread, but the C memory model doesn't necessarily require that. So you're
in danger of the optimiser re-writing the code into an infinite loop.

However, I don't really care enough that I'll fix it myself, so I'll just close the issue.

closed 2021-06-22 by m8pple

DrongoTheDog, 2021-06-22:
> *CAUTION:* This e-mail originated outside the University of Southampton.
> 
> I think you're assuming that writes from one thread will automatically 
> appear on
> another thread, but the C memory model doesn't necessarily require that. 
> So you're
> in danger of the optimiser re-writing the code into an infinite loop.

I didn't know that. Seriously? Explain.....

> However, I don't really care enough that I'll fix it myself, so I'll
> just close the issue. [...]