POETSII/Orchestrator #257: handler_log is treated in the opposite direction

m8pple, 2021-06-23:
For handler_log(level) the API has it that lower levels are more important,
so we include values below the user-specified log-level at compile-time/run-time:

https://github.com/POETSII/poets_improvement_proposals/blob/2a355282f3aa3912bdacf4ec50bf25199bdf310d/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L129-L138

In the orchestrator this is flipped, so it prints messages where the level is higher
than the user-specified level:

https://github.com/POETSII/Orchestrator/blob/fd69625bf05313e163d848bb60c0aa7c216a312c/Source/Softswitch/inc/softswitch.h#L37-L43

heliosfa, 2021-06-23:
Has this changed between schema versions? I recall it being explained to me (possibly incorrectly) that higher levels were more important to give app designers more flexibility.

In any case, it is a simple change that I am happy to make.

m8pple, 2021-06-23:
I don't think it was properly documented till v4, as it was not mentioned in v3 or earlier. Alex Rast (I think?)
correctly argued that it should be in the schema if it was an ambient function that needed special
behaviour from the implementation.

So it's always been that way round in v1, v2, v3, but was only every implicitly documented by the behaviour
of POETS-ecosystem in hardware, and the various simulators in software.

heliosfa, 2021-06-23:
yep, this is one of those places where our implementation differs. Ours (back to v2) has always had it the opposite way round. I'll PR your commit so that we end up doing the same thing.

PR #263 (2021-06-23): Invert the sense of handler-log. 
Fixes #257 so that we are consistent with the v4 spec and other implementations.
closed 2021-06-23 by heliosfa