POETSII/Orchestrator #262: Allow P_CNC_KILL packets to interact with the Softswitch barrier

mvousden, 2021-06-23:
Small discussion arose in https://github.com/POETSII/orchestrator-documentation/pull/18.

The motivation is to make the Mothership able to recall initialised applications without first starting and stopping them.

heliosfa, 2021-06-23:
On further thought, I am in two minds on this. 

It adds a useful feature, but it does mean that if we somehow end up with rogue kill packets floating about during the barrier, they could derail an app execution rather than being silently dropped as now. Also need to consider instruction overhead.

mvousden, 2021-06-23:
I think it might be more important to know why there are rogue kill packets, rather than ignoring them.

Instruction space is a reasonable concern. It might be better for the Mothership to start and stop the application to achieve a recall. Then again, the operator might not want to start the application because they've just realised their supervisor would do something nasty on startup. In that case, the start-stop approach wouldn't solve the problem.

heliosfa, 2021-11-02:
We are going to need to think about how this will interact with the proposed fix for #285.