POETSII/Orchestrator #290: Orchestrator SegFaults when making use of automated exit

heliosfa, 2021-11-01:
**Preamble**
This definitely applies to automated exiting (where `exit /at = "stop"` is used in the call file and the `Super::stop_application();` API call is used) but may also apply where a manual exit is triggered.

This occurs on the `BUGFIX-0285-HardwareIdleBarrier` branch with `FEATURE-0242-HardwareIdle-Mothership` merged in locally. It may occur on other branches.

It occurs on both Ayres and Byron.

**Issue**
In approximately 10% of the runs I have been doing as part of the data gathering for the hardware paper, I have ended up with a seg fault after the application has completed. This is a separate seg fault to the other one that I am about to log.

Example application [calc.xml](https://github.com/POETSII/Orchestrator/files/7455507/calc.xml.txt) and call file [calc.poets](https://github.com/POETSII/Orchestrator/files/7455508/calc.poets.txt) (remove .txt from both)

Example output with the seg fault:

```
gmb@byron:~/Orchestrator$ ./orchestrate.sh -b calc.poets
POETS> 18:39:26.01:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p0.plog'.
POETS> 18:39:26.01: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS> 18:39:26.01:  20(I) The microlog for the command 'call /file = "/home/gmb/Orchestrator/calc.poets"' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p1.plog'.
POETS> 18:39:26.06:  20(I) The microlog for the command 'exit /at = stop' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p2.plog'.
POETS> 18:39:26.06:  69(I) Exit trigger staged - will exit when the next application is completely stopped.
POETS> 18:39:26.06:  20(I) The microlog for the command 'load /engine = ../Tests/StaticResources/Dialect3/Valid/8_box.uif' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p3.plog'.
POETS> 18:39:26.06: 134(I) Clearing internal hardware topology representation O_.POETSHardwareOneBox.ocfg
POETS> 18:39:26.06: 140(I) Topology loaded from file ||../Tests/StaticResources/Dialect3/Valid/8_box.uif||.
POETS> 18:39:26.06:  20(I) The microlog for the command 'load /app = +calc.xml' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p4.plog'.
POETS> 18:39:26.06: 235(I) Application file ../calc.xml loading...
POETS> 18:39:26.06:  65(I) Application file ../calc.xml loaded in 63 ms.
POETS> 18:39:26.06:  20(I) The microlog for the command 'tlink /app = *' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p5.plog'.
POETS> 18:39:26.06: 234(I) Typelinking graph instance 'calc'...
POETS> 18:39:26.06: 249(I) Successfully typelinked graph instance 'calc'.
POETS> 18:39:26.07:  20(I) The microlog for the command 'place /spread = *' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p6.plog'.
POETS> 18:39:26.07: 309(I) Attempting to place graph instance 'calc' using the 'spre' method...
POETS> 18:39:26.07: 302(I) Graph instance 'calc' placed successfully.
POETS> 18:39:28.08:  20(I) The microlog for the command 'compose /app = *' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_26p7.plog'.
POETS> 18:39:28.08: 803(I) Composing graph instance 'calc'...
POETS> 18:39:28.08: 804(I) Graph instance 'calc' composed successfully.
POETS> 18:39:28.09:  20(I) The microlog for the command 'deploy /app = *' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_28p0.plog'.
POETS> 18:39:28.09: 184(I) Deployment of graph instance 'calc' staged. Waiting for Mothership(s) to acknowledge receipt in the background.
POETS> 18:39:28.09:  20(I) The microlog for the command 'initialise /app = *' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_28p1.plog'.
POETS> 18:39:28.09: 187(I) Initialisation of graph instance 'calc' staged. Waiting for Mothership(s) to acknowledge receipt in the background.
POETS> 18:39:28.09:  20(I) The microlog for the command 'run /app = *' will be written to '../Output/Microlog/Microlog_2021_11_01T18_39_28p2.plog'.
POETS> 18:39:28.09: 188(I) Run of graph instance 'calc' staged. Waiting for Mothership(s) to acknowledge receipt in the background.
POETS> 18:39:43.43: 529(I) Mothership (rank 2): Deployment of application 'Arith::calc' (to this Mothership) complete.
POETS> 18:39:43.43: 530(I) Mothership (rank 2): Initialising fully-defined application 'Arith::calc'.
POETS> 18:39:43.43: 186(I) Application 'Arith::calc' successfully deployed on all Motherships it is mapped to.
POETS> 18:39:43.57: 531(I) Mothership (rank 2): Initialisation of application 'Arith::calc' (to this Mothership) complete.
POETS> 18:39:43.57: 186(I) Application 'Arith::calc' ready to start on all Motherships it is mapped to.
POETS> 18:39:43.57: 532(I) Mothership (rank 2): Starting (running) fully-initialised application 'Arith::calc'.
POETS> 18:39:43.57: 186(I) Application 'Arith::calc' running on all Motherships it is mapped to.
POETS>PROBLEM COMPLETE in 12 seconds (11.4631s)
 18:39:55.96: 526(I) Mothership: Supervisor for application 'Arith::calc' has requested it to be stopped.
POETS> 18:39:55.96: 189(I) Stop of graph instance 'calc' staged. Waiting for Mothership(s) to acknowledge receipt in the background.
POETS> 18:39:55.96: 533(I) Mothership (rank 2): Stopping application 'Arith::calc' (which has been started).
POETS> 18:39:55.96: 186(I) Application 'Arith::calc' stopped on all Motherships it is mapped to.
POETS> 18:39:55.96:  68(I) Exit trigger hit - closing down the Orchestrator. The prompt is disabled.
POETS>gmb@byron:~/Orchestrator$
gmb@byron:~/Orchestrator$
gmb@byron:~/Orchestrator$
gmb@byron:~/Orchestrator$
gmb@byron:~/Orchestrator$ ./orchestrate.sh -b calc.poets
POETS> 18:43:46.01:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2021_11_01T18_43_46p0.plog'.
POETS> 18:43:46.01: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS> 18:43:46.01:  20(I) The microlog for the command 'call /file = "/home/gmb/Orchestrator/calc.poets"' will be written to '../Output/Microlog/Microlog_2021_11_01T18_43_46p1.plog'.
POETS> 18:43:46.01:  20(I) The microlog for the command 'exit /at = stop' will be written to '../Output/Microlog/Microlog_2021_11_01T18_43_46p2.plog'.
POETS> 18:43:46.01:  69(I) Exit trigger staged - will exit when the next application is completely stopped.
POETS> 18:43:46.06:  20(I) The microlog for the command 'load /engine = ../Tests/StaticResources/Dialect3/Valid/8_box.uif' will be written to '../Output/Microlog/Microlog_2021_11_01T18_43_46p3.plog'.
POETS> 18:43:46.06: 134(I) Clearing internal hardware topology representation O_.POETSHardwareOneBox.ocfg
POETS> 18:43:46.06: 140(I) Topology loaded from file ||../Tests/StaticResources/Dialect3/Valid/8_box.uif||.
POETS> 18:43:46.06:  20(I) The microlog for the command 'load /app = +calc.xml' will be written to '../Output/Microlog/Microlog_2021_11_01T18_43_46p4.plog'.
POETS> 18:43:46.06: 235(I) Application file ../calc.xml loading...
[... 15 lines cut ...]
POETS> 18:43:48.09: 188(I) Run of graph instance 'calc' staged. Waiting for Mothership(s) to acknowledge receipt in the background.
POETS> 18:44:03.40: 529(I) Mothership (rank 2): Deployment of application 'Arith::calc' (to this Mothership) complete.
POETS> 18:44:03.40: 530(I) Mothership (rank 2): Initialising fully-defined application 'Arith::calc'.
POETS> 18:44:03.40: 186(I) Application 'Arith::calc' successfully deployed on all Motherships it is mapped to.
POETS> 18:44:03.53: 531(I) Mothership (rank 2): Initialisation of application 'Arith::calc' (to this Mothership) complete.
POETS> 18:44:03.53: 532(I) Mothership (rank 2): Starting (running) fully-initialised application 'Arith::calc'.
POETS> 18:44:03.53: 186(I) Application 'Arith::calc' ready to start on all Motherships it is mapped to.
POETS> 18:44:03.53: 186(I) Application 'Arith::calc' running on all Motherships it is mapped to.
POETS>PROBLEM COMPLETE in 11 seconds (11.4583s)
 18:44:14.92: 526(I) Mothership: Supervisor for application 'Arith::calc' has requested it to be stopped.
POETS> 18:44:14.92: 189(I) Stop of graph instance 'calc' staged. Waiting for Mothership(s) to acknowledge receipt in the background.
POETS> 18:44:14.93: 533(I) Mothership (rank 2): Stopping application 'Arith::calc' (which has been started).
POETS>
===================================================================================
=   BAD TERMINATION OF ONE OF YOUR APPLICATION PROCESSES
=   PID 24925 RUNNING AT byron
=   EXIT CODE: 139
=   CLEANING UP REMAINING PROCESSES
=   YOU CAN IGNORE THE BELOW CLEANUP MESSAGES
===================================================================================
YOUR APPLICATION TERMINATED WITH THE EXIT STRING: Segmentation fault (signal 11)
This typically refers to a problem with your application.
Please see the FAQ page for debugging suggestions
gmb@byron:~/Orchestrator$
```

mvousden, 2021-11-03:
Note for later - thinking this might be due to an inconsistent interaction between the scheduled exits, and non-blocking buffered-sent messages in MPI.

heliosfa, 2021-11-03:
> Note for later - thinking this might be due to an inconsistent interaction between the scheduled exits, and non-blocking buffered-sent messages in MPI.

That makes sense, but not in the case where I have had similar seg faults from a "manual" exit as well.

Also, it is nowhere near as prevalent on Byron. out of ~90 runs on Byron, I have had maybe three seg faults of the above pattern.

mvousden, 2021-12-09:
I've confirmed that the Mothership is causing the segmentation fault. Specifically, the segfault consistently triggers at `SuperDB.cpp:119`, though this may not be the actual cause, because the Mothership is multithreaded.

The investigation continues...

closed 2022-03-28 by mvousden