POETSII/Orchestrator #294: Orchestrator does not support running on a single box in the 8-box cluster that is NOT Byron

heliosfa, 2021-11-05:
When attempting to start the Orchestrator on a single box in the 8-box cluster that is not Byron (e.g. if you want to do a run on Defoe because someone is using Byron/Coleridge), Debuglink prints `Requested box sub-mesh of size 2x4 is not valid from box defoe`. This occurs when using the default `POETSHardwareOneBox.ocfg`.

Digging into it, it is because the [Mothership defaults](https://github.com/POETSII/Orchestrator/blob/b2534f390ca532a65e8d35e36173587a138ebe5b/Source/Mothership/Mothership.cpp#L98) to using the environment variables of `HOSTLINK_BOXES_X` and `HOSTLINK_BOXES_Y` or the `TinselBoxMeshXLen` and `TinselBoxMeshYLen` from TInsel's `config.h` to set the size of sub-mesh requested. On all of the boxes on the 8-box cluster, this obviously reflects the size of the full 8-box cluster.

Why is this a problem? Well, it means that if you want a single-box run, you HAVE to use Ayres. It also means that if you want to do a single-box run on Byron, you unintentionally take over the whole cluster. The same is true for 2-box, 4-box or 6-box runs: we can only currently initiate them from Byron and it takes over the whole cluster. This is not very cash money...

How to fix? I guess there are a few options, but given the capabilities of Tinsel, I am wondering whether providing a launch-time argument to `orchestrate.sh` (maybe a `/s` argument...) and/or adding an entry to the config that sets the sub-mesh size at runtime is the way forward. [Tinsel supports](https://github.com/POETSII/tinsel/blob/13b6a7aa1c9ec6a2c8dcf7a39aa61b619de3bd3b/hostlink/DebugLink.cpp#L87) using a sub-mesh of the boxes. We just have to ask it for something smaller.

mvousden, 2021-11-05:
Launcher/HDF integration? The comments of `Mothership::load_backend` show that I was quite annoyed by this.

heliosfa, 2021-11-05:
I was thinking having an optional config option for "number of boxes" (default is to use the env/config.h values) with the ability to override at orchestrate/launcher to set something smaller.

PR #296 (2021-11-16): Permit cluster size definition via the Launcher
This changeset resolves #294 by adding two CLI switches to the launcher (`x` and `y`) to allow the operator to configure the Tinsel-facing cluster shape. When any of these switches are used, the values of environment variables `HOSTLINK_BOXES_X` and `HOSTLINK_BOXES_Y` read by the Mothership (and Hostlink) are changed to the arguments passed in. The launcher will warn if these are not integers (which Tinsel mandates), but will allow the operator to continue regardless (Hostlink will vomit down the line).

Note that this will not stop users from lo [... cut]
closed 2021-11-16 by mvousden