POETSII/Orchestrator #300: XML parser error on attribute contents

m8pple, 2021-12-25:
```
========================================================================================================================
25/12/2021 10:32:32.12 file ../Output/Microlog/Microlog_2021_12_25T10_32_32p0.plog
command [load /app = "/home/dt10/poets-dpd/.test/test_staticWaterSmall/in.xml"]
from console
========================================================================================================================

------------------------------------------------------------------------------------------------------------------------
25/12/2021 10:32:32.12

Checking client file /home/dt10/poets-dpd/.test/test_staticWaterSmall/in.xml

   (lin,col) refers to element closure location in client file

Syntax analysis...

==========================================================
XML syntax error
Error code             : (2) Unexpected token encountered
Line, col              : 2475,7
(Last) symbol token    : {
Lexer token history....
(Line,Col)
(2475,  7) {      <- Token out of sequence
(2475,  6) '
(2475,  5) =
(2475,  4) P
(2474, 65) \n
(2474, 64) '
(2474, 63) basic_dpd_engine_v5_r0
(2474, 41) '
(2474, 40) =
(2474, 39) graphTypeId
```

m8pple, 2021-12-25:
Reduced case causing error
```
<?xml version='1.0' encoding='ASCII'?>
<Graphs xmlns="https://poets-project.org/schemas/virtual-graph-schema-v4" formatMinorVersion="0">
  <GraphType id="basic_dpd_engine_v5_r0">
    <Properties/>
    <SharedCode />
    <MessageTypes />
    <DeviceTypes />
</GraphType>
<GraphInstance id='blurble' graphTypeId='basic_dpd_engine_v5_r0'
   P='{}' >
  <DeviceInstances />
  <EdgeInstances />
</GraphInstance>

</Graphs>
```

Parser error:
```
========================================================================================================================
25/12/2021 10:55:06.59 file ../Output/Microlog/Microlog_2021_12_25T10_55_06p0.plog
command [load /app = "/home/dt10/poets-dpd/.test/test_staticWaterSmall/in.xml"]
from console
========================================================================================================================

------------------------------------------------------------------------------------------------------------------------
25/12/2021 10:55:06.59

Checking client file /home/dt10/poets-dpd/.test/test_staticWaterSmall/in.xml

   (lin,col) refers to element closure location in client file

Syntax analysis...

==========================================================
XML syntax error
Error code             : (2) Unexpected token encountered
Line, col              : 10,7
(Last) symbol token    : {
Lexer token history....
(Line,Col)
(  10,  7) {      <- Token out of sequence
(  10,  6) '
(  10,  5) =
(  10,  4) P
(   9, 65) \n
(   9, 64) '
(   9, 63) basic_dpd_engine_v5_r0
(   9, 41) '
(   9, 40) =
(   9, 39) graphTypeId
```

m8pple, 2021-12-25:
XML parser doesn't seem to support single quote attributes, or supports them only in some cases.

Changing:
```
<GraphInstance id='blurble' graphTypeId='basic_dpd_engine_v5_r0' P='{}'>
```
to:
```
<GraphInstance id='blurble' graphTypeId='basic_dpd_engine_v5_r0' P="{}">
```
seems to let it parse, even though the other attributes still use single quotes.

m8pple, 2021-12-25:
Also happens on EdgeI attributes, .e.g:
```
 <EdgeInstances>
   <EdgeI path='c_3_3_3:migrate_in-c_0_0_0:migrate_out' />
```

Gets the error:
```
25/12/2021 11:10:51.14

Checking client file /home/dt10/poets-dpd/.test/test_dimersSmall/in.xml

   (lin,col) refers to element closure location in client file

Syntax analysis...

==========================================================
XML syntax error
Error code             : (2) Unexpected token encountered
Line, col              : 2544,24
(Last) symbol token    : :
Lexer token history....
(Line,Col)
(2544, 24) :      <- Token out of sequence
(2544, 23) c_3_3_3
(2544, 16) '
(2544, 15) =
(2544, 14) path
(2544,  9) EdgeI
(2544,  4) <
(2543, 18) \n
(2543, 17) >
(2543, 16) EdgeInstances
```

DrongoTheDog, 2022-01-27:
This needs looking at. According to my notes, I fixed this (i.e. included single quotes as valid delimiters for attributes) and issue 307 (compound names) on 18/6/2019. Because I am a ageing recidivist reactionary, I would simply have lobbed it at MLV/GMB and let them get on with it. Further research is needed.