POETSII/Orchestrator #306: Examples of 1..8-box hardware configuration files

m8pple, 2022-01-10:
When trying to use all boxes from byron, I couldn't find any hardware configuration
files for it. The configurations I found were:
- Default 1 box: https://github.com/POETSII/Orchestrator/blob/development/Config/POETSHardwareOneBox.ocfg
- Locally installed 8 box config on byron: `/local/orchestrator-common/8_box.uif`

If I try to load 8_box.uif on byron, using e3a76009322a1b5f1a2ace887eebcb56a768a4c8, I get:
```
dt10@byron:~/poets-dpd/experiments/orch-scaling$ ~/Orchestrator/orchestrate.sh 
POETS> 20:05:50.02:  20(I) The microlog for the command 'load /engine = "../Config/POETSHardwareOneBox.ocfg"' will be written to '../Output/Microlog/Microlog_2022_01_10T20_05_50p0.plog'.
POETS> 20:05:50.02: 140(I) Topology loaded from file ||../Config/POETSHardwareOneBox.ocfg||.
POETS>load /engine = "/local/orchestrator-common/8_box.uif"
POETS> 20:08:27.04:  23(I) load /engine = "/local/orchestrator-common/8_box.uif"
POETS> 20:08:27.04:  20(I) The microlog for the command 'load /engine = "/local/orchestrator-common/8_box.uif"' will be written to '../Output/Microlog/Microlog_2022_01_10T20_08_27p0.plog'.
POETS> 20:08:27.04: 134(I) Clearing internal hardware topology representation O_.POETSHardwareOneBox.ocfg
POETS> 20:08:27.04: 141(E) Errors encountered loading hardware input file at ||/local/orchestrator-common/8_box.uif||:
Error(s) were encountered while loading the hardware description file '/local/orchestrator-common/8_box.uif'. Since the reader fails slowly, consider addressing the top-most errors first:
 - Variable 'pair_cores' not defined in the 'mailbox(CommonMbox)' section.
 - Variable 'pair_cores' not defined in the 'mailbox(CommonMbox)' section.
 - Variable 'pair_cores' not defined in the 'mailbox(CommonMbox)' section.
 - Variable 'pair_cores' not defined in the 'mailbox(CommonMbox)' section.
 - Variable 'pair_cores' not defined in the 'mailbox(CommonMbox)' section.
 - Variable 'pair_cores' not defined in the 'mailbox(CommonMbox)' section.
```
plus lots more `pair_cores` errors.

There is documentation of the different dialects in Orchestrator-documentation, but I can't find any
actual configuration files for 8 box (or other size) configurations.

heliosfa, 2022-01-10:
Checking my generators, I have been using the one at `Tests/StaticResources/Dialect3/Valid/8_box.uif` for my 8-box runs.

I have a feeling that the one at `/local/orchestrator-common/8_box.uif` is outdated. I will put it on my list to update.

m8pple, 2022-01-12:
Thanks that worked.

It would be useful to have configs for all 1,2,4, and 8 boxes included in the standard Orchestrator alongside the
one box config for scaling experiments. Or in orchestrator-common.

closed 2022-01-12 by m8pple