POETSII/Orchestrator #308: RTS_FLAG_{device_type}_{name} not emitted

m8pple, 2022-01-13:
The RTS_FLAG_x values have both a device local and a global version, so that they
can be referred to from both device and global shared code.

https://github.com/POETSII/poets_improvement_proposals/blob/438757c772dfa26cecd56b2856c6c2e180308530/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L368-L372

This is used in a few applications, e.g. "clock_tree_broad" from the 2019-09-06 benchmarks. I don't
think the values are are emitted by composer, or at least they aren't visible in scope for the handlers.

It can be worked around using sed, but makes it a bit more complex to pass the benchmark set through the
Orchestrator.

heliosfa, 2022-01-13:
Currently, the rts bitfield is a purely local construct in `softswitch_onRTS()` ([softswitch_common.cpp:710](https://github.com/POETSII/Orchestrator/blob/e74e6ee1e353935e074e4f9a522020aa029f9351/Source/Softswitch/src/softswitch_common.cpp#L710)).

This could be modified to make it persistent, but obviously comes with overhead as it needs to be stored for each device. Are these meant to be pointers or straight up variables?

I'm just trying to get my head around the need for this functionality?

Also, I am probably being dense but I don't see the need for both (or the difference between) `RTS_FLAG_{device_type}_{name}` and `RTS_INDEX_{device_type}_{name}`, especially as "requestIndex will be contiguous and start at zero so they can be used as bit-field masks" ?

heliosfa, 2022-01-13:
Clarification: this is about changing the scope of `RTS_FLAG_X` and `RTS_INDEX_X` consts emitted as part of the build.

They are currently emitted as `RTS_FLAG_<pinname>` and `RTS_INDEX_<pinname>` as part of the per-core handlers header rather than with the device-type specifier.

The way forward: we can dump them into one of the shared headers (possibly DeviceStructs.h) with the correct name(s) by adding an iteration over the device types (and the pins they have) pre per-core generation