POETSII/Orchestrator #309: Conditional send through doSend is not supported

m8pple, 2022-01-13:
Some applications request the opportunity to send, but then can choose not to:

https://github.com/POETSII/poets_improvement_proposals/blob/438757c772dfa26cecd56b2856c6c2e180308530/proposed/PIP-0020/virtual-graph-schema-v4.rnc#L500

This is sometimes used when it is expensive to work out when to send (essentially you have to construct the whole
packet first), or if you want to do something based on time, so you don't want to do it in the RTS handler. I think
Shane also used it in his firefly stuff.

It's also used in the clock_tree_broad and deep apps from the benchmark/stress tests. Probably some others too,
I haven't looked at them all. The doSend flag pre-dates OnCompute, which pre-dates OnDeviceIdle, which would
cover some use-cases these days.

m8pple, 2022-01-13:
Also used in the "airfoil" finite element app from the stress tests. Oh well.

heliosfa, 2022-01-13:
I recall looking at this before after we talked about it and thinking it would be  "quick win" but then remembered that the buffering mode ADB requested complicated matters. This should not be too difficult to implement once I am off papers.